Digital-Naturalism-Laboratories/Mothbox #10: 3.1 Schedule not working

Hubertszcz, 2024-04-07:
I set up MothBox02 the past two nights and the scheduling was not what it should have been. I was expecting three photos (for HDR) every two minutes, for one hour starting at 7:30, 10:30, 1:30, and 4:30.  

First off, 3 photos were taken every minute.

I started it manually at 6:25pm on the 4th of April, and it took photos every minute during the following time blocks, according to the file names (the file names are different from the photo metadata times, which I think are in UTC, another issue):

18:25 - 19:23
19:31 - 21:27     ----- this is strange, ran for two hours. A bunch of the photos are red because it was taking them while in my backpack between deployment sites.
22:31 - 23:29
[now the 5th]
1:31 - 2:29
4:31 - 5:29

19:31 - 20:29
22:31 - 23:29
[Now the 6th]
1:31 - 2:29
4:31 - 5:28

Seems alright, except when I picked up the MothBox at around 11:30 it was still running. You can see that the last photos taken on the 6th are black (in my backpack), and before that it's daylight. 

Something went wrong with the clock between the morning and evening of the 5th, while it was deployed in the forest?

I would also be useful to automatically generate a CSV listing the file name, time, and folder of all the photos produced, and a summary of the time blocks those photos represent. There can be a script to update the csv based on the photos in the nightly saved folder, at the end of every night.

quitmeyer, 2024-04-09:
this is odd, let's look at it some more and see

quitmeyer, 2024-06-28:
schedules should work fine and tested in pi4 or pi5

closed 2024-06-28 by quitmeyer