Digital-Naturalism-Laboratories/Mothbox #11: HDR exposure range needs to be narrowed down

Hubertszcz, 2024-04-07:
The 3rd photo of every set (underexposed) is way too underexposed - none of the insects are usably exposed. 
The 2nd photo of every set appears too overexposed - insects have indistinct edges because they're overpowered by the brightness of the sheet. This one I'm less sure about - maybe a large dark moth would need this level of exposure.

quitmeyer, 2024-04-07:
In the camera settings csv you can adjust the hdr width parameter

https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/Software/camera_settings.csv

Hubertszcz, 2024-04-07:
Are the times in the schedule app true times on-location, or are the UTC times? 

If I want 7:30pm Panama, do I schedule 12:30?

quitmeyer, 2024-04-07:
You put in the location and the gmt offset

Its default is -5 for Panama.

All the metadata and name of the photos has always been in Panama time for
me.

If it isn't starting up at the correct times, it maybe lost the power to
the pijuice battery? Which would lose the rtc clock

But if it is running at the correct times, it should be listing the correct
times on the files. Some computers might originally read file date metadata
as a weird offset, but eventually figures itself out

Odd if the files are way off

closed 2024-06-28 by quitmeyer