Digital-Naturalism-Laboratories/Mothbox #30: [FEATURE] Power consumption estimates

mihow, 2024-12-13:
Do you have a breakdown of the power consumption for the recommended components?

The [docs suggest ](https://digital-naturalism-laboratories.github.io/Mothbox/docs/building/powersources/) that the 38,400mAh Talentcell battery (PB120B1) should be able to power the Mothbox for 22 hours (4 nights at 5.5hrs each)

Is that with the Mothbeam light? And 2 photo ring lights firing intermittently? 

If you have a table of values you use to estimate the battery size that would be fantastic!

Here are my estimates so far just from googling
[image]

Thanks for all your great research and effort devoted to this project!

quitmeyer, 2024-12-17:
cool, we have some in the background ill put up on the documentation shortly.  We didn't actually calculate the 22 hours, that's from just running many of them over and over again with a previous mothbeam (mothbeam v1) hooked up. We have a newer mothbeam now that's quite a bit brighter on the UV pcb, but also uses about twice as much juice. So the current runtime with mothbeam v2 is about 12-13 hours. 

The amp-hour estimate (38400mAh) on the PB120b1 seems off or i don't understand something. but their watt hours seems to agree with my reality ( 142.08Wh ):

here's a quick rundown of the different power states. These come from some measurements hooked up to the 12V battery:

**1.  during off state (when everything is pretty much completely off)**
~1ma = .012 watts

Normal pi5 shutdown mode is like 1.3 watts
But we shift ours for ultra standby which is only like .05 watts (which i guess at 5v would still be 1ma)

**2.  when the attractors are on**
each mothbeam seems to take like 300ma  = 4 watts
so like 
8 watts for our standard config with a UV and Visible mothbeam

plus the pi is generally idle except for the like couple seconds it is taking a photo. During idle time it seems to consume about 2.4 watts (wifi and bluetooth and stuff are shut off). When taking a photo this can jump up briefly, but like super short.

but the power consumption around this time seems to be around 
10.4 watts total

for the ideal PB120b1 at 142 watt hours, that suggests this would run for 13.65 hours (which seems to agree with our latest testing of the more power hungry v2 mothbeams)

the reality is more like 12-13 hours, so i attribute that extra power loss to the ring flashes and non-ideal battery world.

3-  when it is taking pictures and the flash is on as well.**
The photo taking takes less than half a second usually and doesn't seem to significantly impact the overall battery life as much as the other stuff (like 45 minutes less out of 13 hours predicted, or about 6% energy loss)

ToonVanEyck, 2025-06-05:
I saw you are connecting all your power LED's in parallel. 

[image]
If you instead create 5 parallel branches of 2 series LED's with a resistor, you could lower the amount of power a bit. You are wasting a lot of power in the resistors at the moment.

quitmeyer, 2025-06-05:
wonderful!