Digital-Naturalism-Laboratories/Mothbox #32: Pullup Rs on dip-switches

biomurph, 2025-06-05:
Question about using pullup resistors on the dip switch UI.
Those switches go to multiplexers, and the multiplexers are 'break-before-make', which makes one think that the pullups are not necessary. Also, given the enable pins on the muxers, they can be turned 'off' to break the connection from the Mux pins to the I/O common pin.

However, the dip switches are either open or closed. See the "Schematics" on the dip switch datasheet
https://www.lcsc.com/datasheet/lcsc_datasheet_2409272202_SM-Switch-SMXS-08K-TP_C5439767.pdf
Given the dip switch circuit, you need the pullups, for sure. Otherwise, shorty will get ya!

If you're concerned about routing issues and board space, take a look at a resistor array like this one
https://eu.mouser.com/ProductDetail/Panasonic/EXB-2HV103JV?qs=oE3VwAv%2FYtnxG1PkyhflFQ%3D%3D

biomurph, 2025-06-05:
The dip switch makes connection or disconnection between its pins. Without the pullups, closing the dip switch would short 5v to GND!

quitmeyer, 2025-06-05:
but the dip switch won't connect to 5V! it will connect GND to the GPIO pin. The GPIO pin will be set by software to have an internal pullup. So when the switch is OPEN the GPIO pin will read HI, and when the switch connects to GND the GPIO pin will read LOW (i think)