Digital-Naturalism-Laboratories/Mothbox #34: Power Trace Vias Are Too Small

biomurph, 2025-06-21:
**Describe the bug**
You’re power trace vias on the UV and LED blasters are too small, I think.
Have you calculated currents?

Here’s a link to understanding more
https://www.pcbrunner.com/optimising-your-pcb-design-for-via-current-handling-a-simple-guide/

quitmeyer, 2025-06-21:
Oh wow that's so interesting! I hadn't even thought about the limitations
of the vias!

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

parkview, 2025-07-15:
in PCB v5.0.1 I kept the via's the same diameter, but added multiple via's.

closed 2025-07-15 by parkview