Digital-Naturalism-Laboratories/Mothbox #54: Found your backup script bug

mochuelo-kun, 2025-07-14:
https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/e8072f6fed8cddc4fd3c084dfdf87616332fdd21/Firmware/Backup_Files.py#L60
-->
Change to:
`return stat.f_blocks * stat.f_frsize, stat.f_bavail * stat.f_frsize`

statvfs attributes are confusing AF. I don't know how

- `unsigned long f_bsize:	The file system block size`
- and `unsigned long f_frsize:	The fundamental file system block size`

are different exactly ("fundamental"??), but the one you want is `f_frsize`, since

- `unsigned long f_blocks:	The total number of blocks on the file system in units of f_frsize`

Tried this on my local computer with both variations, and `f_frsize` gave the correct answer, whereas `f_bsize` gave something wildly too large.

See for example documentation: https://www.ibm.com/docs/en/zos/2.4.0?topic=functions-statvfs-get-file-system-information

Also might want to fix the indentation on this line https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/e8072f6fed8cddc4fd3c084dfdf87616332fdd21/Firmware/Backup_Files.py#L434

Also also, if you want to flip back to using `rsync`, the `-c` option should use a checksum (if your version of rsync has that feature). SOme more info here https://unix.stackexchange.com/questions/298778/is-rsync-the-fastest-and-most-reliable-way-to-verify-whether-directories-are-ide (just `man rsync` for all the extra little options though - there are a ton).

arky, 2025-07-28:
@mochuelo-kun Could you please send PR? Thanks!

quitmeyer, 2025-07-28:
I fixed up already :)

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

closed 2025-08-05 by quitmeyer