Digital-Naturalism-Laboratories/Mothbox #58: Metadata generation in most recent release?

chevalierid, 2025-07-17:
**Describe the bug**
The "create your own metadata" link in the [metadata](https://digital-naturalism-laboratories.github.io/Mothbox/docs/processing/OrganizeData/#metadata) section of the documentation is broken. I would guess it pointed to the metadata generation script (Mothbot_GenMetadata.py) that is now in the "old" folder. Is this script deprecated? If so, is there a new way to generate the JSON files needed for the next steps?

**To Reproduce**
Click the hyperlinked text "create your own metadata" in the [metadata](https://digital-naturalism-laboratories.github.io/Mothbox/docs/processing/OrganizeData/#metadata) section of the documentation.

Alternatively, try to run Mothbot_CreateDataset.py after changing the file paths to match your use case.

**Expected behavior**
404 Not Found appears.

Alternatively, Mothbot_CreateDataset.py will fail to find any samples because it cannot find any sets of three similarly-named grouped files (one JPG, two JSON).

Hope this is OK to post as an issue. Would appreciate any help you can provide!

quitmeyer, 2025-07-29:
Thanks for posting the issue!
The gen metadata script was part of an outdated workflow.

If you can let me know what you are trying to do, i can give tips :)

I'm just getting back from running a bit conference and catching up on
things

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

quitmeyer, 2025-08-05:
Ahhh ok @chevalierid , now that I am back at a proper computer, i think i understand your problem!

The link to the metadata form was just broken.

For your metadata, log your metadata in a spreadsheet like this:
https://docs.google.com/spreadsheets/d/1oN4-wttu4Gz7V0yauhQZgHVQOLTyO0c6XzUNWczOqeI/edit?usp=sharing

and then point your script to this CSV :)

I will update the website with a proper link too!

closed 2025-08-05 by quitmeyer

quitmeyer, 2025-08-06:
This example of a CSV works good too

https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/AI/Auto%20Calculations%20-%20Mothbox%20Main%20Metadata%20field%20sheet%20(Bilingue)%20(Responses)%20-%20Form%20responses%201.csv