Digital-Naturalism-Laboratories/Mothbox #60: convert to csv not working

Hubertszcz, 2025-07-28:
Running the Mothbot_ConvertDatasettoCSV script gives the following error:

Found fiftyone dataset's json path
['', '', '', '', '', '', '']No valid deepest rank and value found. Probably ID'ed as an Error category, don't worry
['', '', '', '', '', '', '']No valid deepest rank and value found. Probably ID'ed as an Error category, don't worry
['', '', '', '', '', '', '']No valid deepest rank and value found. Probably ID'ed as an Error category, don't worry
['', '', '', '', '', '', '']No valid deepest rank and value found. Probably ID'ed as an Error category, don't worry
['', '', '', '', '', '', '']No valid deepest rank and value found. Probably ID'ed as an Error category, don't worry
Traceback (most recent call last):
  File "C:\Users\Hubert\AppData\Local\Programs\Python\Python311\Lib\site-packages\pandas\core\indexes\base.py", line 3805, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "index.pyx", line 167, in pandas._libs.index.IndexEngine.get_loc
  File "index.pyx", line 196, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\\_libs\\hashtable_class_helper.pxi", line 7081, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas\\_libs\\hashtable_class_helper.pxi", line 7089, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'taxonRank'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\Users\Hubert\Desktop\Biodiversity Monitoring\MothBox\Mothbox Github\Mothbox\AI\Mothbot_ConvertDatasettoCSV.py", line 359, in <module>
    json_to_csv(INPUT_PATH, UTC_OFFSET, TAXA_LIST_PATH)
  File "c:\Users\Hubert\Desktop\Biodiversity Monitoring\MothBox\Mothbox Github\Mothbox\AI\Mothbot_ConvertDatasettoCSV.py", line 295, in json_to_csv
    taxon_id, scientificName= find_deepest_taxon_info(TAXA, taxa_values, taxa_lookup)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Hubert\Desktop\Biodiversity Monitoring\MothBox\Mothbox Github\Mothbox\AI\Mothbot_ConvertDatasettoCSV.py", line 150, in find_deepest_taxon_info
    filtered_rows = taxa_lookup[taxa_lookup['taxonRank'].str.lower() == deepest_rank.lower()]
                                ~~~~~~~~~~~^^^^^^^^^^^^^
  File "C:\Users\Hubert\AppData\Local\Programs\Python\Python311\Lib\site-packages\pandas\core\frame.py", line 4102, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Hubert\AppData\Local\Programs\Python\Python311\Lib\site-packages\pandas\core\indexes\base.py", line 3812, in get_loc
    raise KeyError(key) from err
KeyError: 'taxonRank'

I initially thought this was an issue with morphospecies IDs from fiftyone, but it doesn't work on files ID'd to order-level that it worked on before. I recently synced all my scripts with github, so perhaps the script was changed since my last sync.

Hubertszcz, 2025-07-28:
[metadata.json](https://github.com/user-attachments/files/21475837/metadata.json)
[samples.json](https://github.com/user-attachments/files/21475838/samples.json)

Here is the data I am trying to turn into a csv

Hubertszcz, 2025-07-28:
[metadata.json](https://github.com/user-attachments/files/21475863/metadata.json)
[samples.json](https://github.com/user-attachments/files/21475862/samples.json)

Here is the order-level version of the same data, which I successfully turned into a csv in the past, but can no longer.

Hubertszcz, 2025-07-28:
[SpeciesList_CountryPanama_TaxaInsecta_Morphospecies.csv](https://github.com/user-attachments/files/21477141/SpeciesList_CountryPanama_TaxaInsecta_Morphospecies.csv)

Panama insects species list augmented with possible morphospecies.

quitmeyer, 2025-08-06:
@Hubertszcz  working on this a bit now. Added more description to the title.

Problem:
The key challenge here is that if we add cool new taxonomies that lie outside of the species list, we get errors. The reason we usually rely on the species list is because to make the CSV darwinCORE compatible, we need the special taxaCodes they use. You also want to be able to add kinda more rare taxonomy like "TRIBE" which GBIF doesn't have.

We also have non-taxonomic tags like "ERROR"

Possible Solutions:
Perhaps a solution to this would be to add checks to try to go as deep in a taxonomic rank as we have values for in an official species list (thus excluding morphospecies) and use that to assign the darwinCORE taxon code while still allowing the morphospecies tags

quitmeyer, 2025-08-06:
I also tried adding numbers to my morphospecies to see if maybe that caused errors?

[image]

but nope, still went ok. 

you can notice that it does leave the taxonID and such blank, and i can work on having that keep the last legit taxa

quitmeyer, 2025-08-06:
It now also adds "ERROR" as a common name if it is marked as an ERROR