Digital-Naturalism-Laboratories/Mothbox #66: Mothbot_CreateDataset.py parameters and libraries

Nova-Scotia, 2025-08-11:
I've successfully run the detect and identify mothbot scripts, so I have a .json file for each photo my mothbox took, and an extra folder named 'patches' in one of my photo days' folders. 

I'm getting stuck at the next step. I've followed along to create an environment in VS code here: [https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/docs/docs/processing/processingsoftware.md#experimental-set-up-a-managed-environment-with-uv](https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/docs/docs/processing/processingsoftware.md#experimental-set-up-a-managed-environment-with-uv)

but there are some places where directions are inconsistent:

1) there seem to be two folders with python scripts in them to analyze data: the Mothbox/AI folder (containing the Mothbot_CreateDataset.py script) and the Moth box/AI/Mothbot folder (which *does not* contain the Mothbot_CreateDataset.py script). Why are there two copies of some scripts, and not others? Will this affect my ability to use the virtual environment if I try to run the script from the AI folder? I've gone back over screenshots from the instructions, and sometimes the scripts are run out of the AI folder, sometimes Mothbot folder. Not sure which is correct.

2) When I try to run the script, it has trouble because I don't think all the libraries in that script are present in the requirements.txt file, such as `piexif`. It's also struggling to find Mothbot_ConvertDatasettoCSV which makes me wonder if there is something wrong with my setup.

3) In the directions for [creating a dataset](https://digital-naturalism-laboratories.github.io/Mothbox/docs/processing/databases/), the first step is creating the thumbnails in the 'patches' folder, but didn't we just do that in the detect script?

4) In the directions for [creating a dataset](https://digital-naturalism-laboratories.github.io/Mothbox/docs/processing/databases/), there is no parameter called 'METADATA_PATH' but in my Mothbot_CreateDataset.py I'm supposed to fill that in. What do I put here? I don't think the previous scripts output a .csv yet?

quitmeyer, 2025-08-11:
Yep i am working on getting these next steps working with the managed UV environment today :)

quitmeyer, 2025-08-11:
basically the folders in the "Mothbot" subfolder are what I am working on getting all working with the managed environment. Hopefully soon all the scripts will be inside there and all the requirements together :)
also putting together a basic UI in there that should be easier to work with!

you are basically on the bleeding edge of what I am coding up, so thanks for testing things out!

quitmeyer, 2025-08-11:
(and also the documentation is always a bit behind what we got working!)

quitmeyer, 2025-08-11:
Ok @Nova-Scotia and @Hubertszcz  and @briannaljohns 
I just successfully ran "Create_dataset" from the new version in /Mothbot
that runs in a virtual environment

[image]

you can try it out!

Some notes:

EXIFTOOL
There is a thing called "ExifTool" that (because exif data on images is super convoluted) needs to be installed to run CreateDataset

if you are using windows, you don't actually have to install it i think. It just points to a relatively positioned folder in the github that has the .exe it runs

[image]

if you are running mac or linux, you need to [download exiftool](https://exiftool.org/)  (or the mac package .pkg is actually in in the AI folder of the github already for you as well)

ADDITIONAL FILES
there are additional files you need like a 
-Species list
-Metadata file

luckily both are already in the AI file, and already relatively linked in the standard create dataset script. You can replace those files with your own files, or just run as is to test it!

Nova-Scotia, 2025-08-12:
Thanks so much for your continued work! We're getting somwhere! I have exiftool already on my windows computer, but downloaded it years ago so not sure if mothbot is finding it.

The script is now available in the Mothbot folder, but I'm still having errors. I don't know how to create that metadata file .csv before trying the CreateDataset script. 

Also getting this when I run the script using your csv already in the AI folder:

```
(Mothbot) PS C:\Users\Owner\Documents\GitHub\Mothbox\AI\Mothbot> & C:\Users\Owner\Documents\GitHub\Mothbox\AI\Mothbot\.venv\Scripts\python.exe c:/Users/Owner/Documents/GitHub/Mothbox/AI/Mothbot/Mothbot_CreateDataset.py        
Traceback (most recent call last):
  File "c:\Users\Owner\Documents\GitHub\Mothbox\AI\Mothbot\Mothbot_CreateDataset.py", line 1055, in <module>     
    metadata= find_csv_match(INPUT_PATH, METADATA_PATH)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\Owner\Documents\GitHub\Mothbox\AI\Mothbot\Mothbot_CreateDataset.py", line 182, in find_csv_match

    raise ValueError("The input path does not contain the expected minimum of 4 parts.")
ValueError: The input path does not contain the expected minimum of 4 parts.
(Mothbot) PS C:\Users\Owner\Documents\GitHub\Mothbox\AI\Mothbot>
```

And some errors (from my interpreter)?

[image]

quitmeyer, 2025-08-12:
There's already a bulk file analyzer script! im working on getting it in the new Mothbot environment

Nova-Scotia, 2025-08-14:
Thanks! I will let you know how it goes when I try! Do I still need to change my default folder structure as you mentioned above?

quitmeyer, 2025-08-14:
Not as much! (At least for the detection and ID)

The create dataset part might still because that's how we connect
deployments and metadata. But that part isn't implemented yet in the UI.

This is just letting you know how to batch process detections and ids.

During the create data set part you'll still have to do those individually
I think for each night.

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

quitmeyer, 2025-08-16:
The Generate CSV tab now also functions in the new Mothbox UI tab as well! It will look for any folders in your list of folders that have had a fiftyone dataset made, and try to create a darwincore CSV for them

quitmeyer, 2025-08-20:
that's awesome @Nova-Scotia !

I think ill close this issue for now, but please log future issues you come across. or if im missing something here, we can reopen