Digital-Naturalism-Laboratories/Mothbox #68: [FEATURE] - Mothbot_Detect skips corrupted photos

Nova-Scotia, 2025-08-21:
I noticed that in mothbot_detect, the script will fail if there are corrupt photos in your directory. This is inefficient when you've set it to run on a lot of photos overnight! It would be great it there was a feature to skip corrupt images and provide maybe a list of them in a .log file afterwards so the user can inspect.

Nova-Scotia, 2025-08-21:
Hm - that's odd! What kind of details would be helpful?

Nova-Scotia, 2025-08-21:
Lenovo P15V Laptop
Processor	12th Gen Intel(R) Core(TM) i7-12700H   2.30 GHz
Installed RAM	32.0 GB (31.7 GB usable)
System type	64-bit operating system, x64-based processor

Windows 11
Virtual Environment
Batch Script (mothbot)

quitmeyer, 2025-08-21:
BTW what is your GPU?

Nova-Scotia, 2025-08-21:
> BTW what is your GPU?
I have two listed, GPU 0 and GPU 1 (exposing my lack of knowledge of such things); Intel(R) Iris(R) Xe Graphics, and NVIDIA T600 Laptop GPU, respectively.

Nova-Scotia, 2025-08-21:
> [@Nova-Scotia](https://github.com/Nova-Scotia) can you do me a favor and try to run the Mothbot_Detect.py script as a standalone script on your folder and see if that works?

I can try but I'm actually not sure how to do that. I'm going on vacation for two weeks after tomorrow so just trying to wrap a few things up - if you give me some quick directions I can, otherwise might have to wait!

Nova-Scotia, 2025-08-21:
Thanks! It works now.