Digital-Naturalism-Laboratories/Mothbox #69: only copying one file

quitmeyer, 2025-08-27:
From Mariana:

We saw that this is now available:
https://digital-naturalism-laboratories.github.io/Mothbox/docs/usage/configure/
THANK YOU!!!!!
Things we figured out:
 Where to setup self-startup times and durations for Raspberry Pi (schedule_setings.csv)
 To "activate" the schedule by calling Scheduler.py
 To call Scheduler.py on Crontab
 To call Atractor_On.py on Crontab
 To call Take_Picture.py on Crontab
Remaining Questions:
1) We used the normal Backup setup that came with the original files (which was being called from crontab) but it is not copying all pictures to the usb stick. It is only copying one from each day. How do we fix this?
2) The single internal blacklight that was recommended but it is a little weak. We instead ziptied an external blacklight (something like this) to the mothbox but got rolling shutter issues. In our own implementation, with the Arducam Hawkeye (which was not great), we figured out we could set a shutter speed of 8333 ms to cancel out the rolling-shutter effects. We tried looking at the file called "camera_settings.csv) but couldn't figure out where to mess with the shutter speed call to libcamera or picamera.

**

quitmeyer, 2025-08-28:
There might be a problem with the new version of shutil, 
https://stackoverflow.com/questions/1868714/how-do-i-copy-an-entire-directory-of-files-into-an-existing-directory-using-pyth

it looks like you need to add a flag now if some of those directories exist. that could be a source of the problem

quitmeyer, 2025-08-29:
I think I found the bug!! I added the flag I mentioned before and it seemed to have solved the one file per night error! 

For right now, to solve this error

Copy all the file contents from this 
https://github.com/Digital-Naturalism-Laboratories/Mothbox/blob/main/Firmware/Backup_Files.py

And replace the backup script that's on the mothbox Raspberry pi! 

I will work on a full new mothbox image that will have this fix built in, and I'll post here when that's ready but for now this is the quickest way to patch this error!

frankpablo, 2025-08-29:
@quitmeyer : Thanks for the feedback! I 

  1. About the Backup:  updated the Backup, which I'll test and report (I'm glad you recommended that change because that's exactly where I was hoping the issue would be)

  2. About Rolling shutter: I had a typo in the message we sent you earlier… I meant: 8.333 ms (or 1/120 of a second). This is because I wanted to try the externally powered blacklight, which has the same wavelength as the one you recommend but needs AC to run (which causes the 60Hz flickering). As you say, it might be better NOT to try to do manual adjustments to the exposure time, so I think the most reasonable thing to do first is to try to run more lights off of the 12V DC battery and see if that does it.