Digital-Naturalism-Laboratories/Mothbox #73: Mothbot_CreateDataset.py does not create files or launch Fiftyone

peetersmark, 2025-09-02:
**Describe the bug**
when running Mothbot_CreateDataset.py, the script ends and does not create the 3 files (one csv and two jason) or load Fiftyone web interface. The images file do have their Exif tags updated, so that part is working.

**To Reproduce**
running Mothbot_CreateDataset.py on win10, in virtual studio code.

**Expected behavior**
3 files should be created in the photo folder and fiftyone web interface should launch

**Additional context**
Attached is the terminal log file, I do see some errors about: 
AttributeError: 'samples.68b742c2d23af69760a23baa' object has no attribute 'ground_height'

[terminal log for Mothbot_CreateDataset-py.txt](https://github.com/user-attachments/files/22102712/terminal.log.for.Mothbot_CreateDataset-py.txt)

quitmeyer, 2025-09-03:
> thanks, i am confused about the metadata file.
> Looking at the page you referenced, the links on the page for "create a
> meta data file" and the PDF samples all seem broken. but i think it is just
> trying to point to the sample CSV file you provide?
>
> *Try1* - Using the sample file you are calling in the script
> '..\Mothbox_Main_Metadata_Field_Sheet_Example - Form responses 1.csv' I
> added a line for me at the top BUT I still got the error: AttributeError:
> 'samples.68b868aa346387d9dbbe8a15' object has no attribute 'ground_height'
>
> *Try2* - I noticed the sample CSV file does not have a header for
> "ground_height", it has a header for "height (placement above ground)", so
> I renamed that in the CSV and ran it again. But I still get the same error.
>
> There are other fields referenced on the Metadata webpage are not in the
> headers of the sample CSV file, so I suspect I am missing something.

Here's what to do:

Get an exact copy of the CSV from the GitHub

Make a new row for your deployment
And fill out all the columns of that row

Here's the tricky part:

To connect your data to
The appropriate row in the csv, it looks to match the deployment name

If you look on that page about organizing data it shows you to put your
data into a set of folders named after the area and point and mothbox name.

So make sure those are formatted the same as your CSV entries and then it
should be able to connect them. (Again working on making this a bit easier
in the future)

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

peetersmark, 2025-09-03:
thanks for this info. I really appreciate the help. I did try to follow the data organization but it seem i have some things wrong. 

**about the folder structure**
Should I NOT use the absolute references in the Mothbot_CreateDataset.py?
My current settings for this file look like this:

> INPUT_PATH = r"D:\mothbox\DATA\kzoo_506BkYardMaple_yearlyBrema_2025-08-31\2025-09-02"
> METADATA_PATH = r'D:\mothbox\DATA\gbif-files\Mark-Mothbox_Main_Metadata-test2.csv'
> UTC_OFFSET= -5 #Panama is -5, Indonesia is 8 change for different locations
> TAXA_LIST_PATH = r"D:\mothbox\DATA\gbif-files\bigMIbox.csv"

following the naming convention I named my deployment folder: kzoo_506BkYardMaple_yearlyBrema_2025-08-31
AND
this was the name I added to the CSV file in last field “deployment name”

this was my folder structure:

> |__kzoo_506BkYardMaple_yearlyBrema_2025-08-31
>        ... |___2025-08-31
>        ... |___2025-09-01
>        ... |___2025-09-02

**SO i think you are saying I need to fix these things?**
1. put the current deployment folder inside parent folder called USA
2. rename the Nightly Folders like this?
| a. 2025-08-31 => 2025-08-01
| b. 2025-09-01 => 2025-08-02
| c. 2025-09-02 => 2025-08-03
3. rename and move the Yolo file like this:
| a. rename yolo11m_4500_imgsz1600_b1_2024-01-18.pt => yearlyBrema_2025-08-31-17-00-00.json
| b. AND move a copy of this into EACH nightly folder
4. rename and more the metadata file like this:
| a. rename Mark-Mothbox_Main_Metadata-test2.csv => yearlyBrema_2025-08-31-17-00-00_metadata.json
| b. AND move a copy of this into EACH nightly folder

quitmeyer, 2025-09-03:
Sorry busy with something but super quick:

Absolute paths looks great!

1 and 2 - yes!

3 and 4 - no! You don't need to change those files or organize the metadata
file or the YOLO file in any special way, they just need to get pointed to
in the script.

By the way have you tried the mothbot_ui.py script for a basic user
interface?

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

quitmeyer, 2025-09-04:
@peetersmark good catch on those deadlinked pdfs!
 i fixed them

also i just fixed the create_dataset script so that if you don't have metadata connected, it will just ignore it! Just tested with a couple batches of randomly named datasets and worked fine, get the latest version of the code off github. so that should solve your problem! ill close this issue, but feel free to keep commenting or open if the same problem comes back.

closed 2025-09-04 by quitmeyer

peetersmark, 2025-09-05:
Thanks the new script is able to open fiftyone! Yeah

BUT what I am seeing is that no metadata is being read. it must be that only the field for ground height was giving an error. 

I did make the changes to the folder structure (1 and 2), and now I am wondering why is it not able to read my metadata CSV file. I must have something wrong but I can see it. attached is the CSV where I have attempted to add a line at the top for my deployment. my data is in row 2. and the file is called with the absolute path like this
METADATA_PATH = r'D:\mothbox\DATA\gbif-files\Mark-Mothbox_Main_Metadata-test2.csv'

Can you see what I am doing wrong?

[Mark-Mothbox_Main_Metadata-test1.csv](https://github.com/user-attachments/files/22162919/Mark-Mothbox_Main_Metadata-test1.csv)

quitmeyer, 2025-09-05:
>
> it must be that only the field for ground height was giving an error.

Nah, that's just the first line where the code starts looking at the
metadata, so it fails first.

 Ill take a look at your stuff and let you know!

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

quitmeyer, 2025-09-05:
> found the prob:
> the deployment name part you got right, but the "area" and "point" were off
> i renamed it for ya! try it!

I also tweaked the code so your original format would have been detected,
and simpified the detecting of matching metadata overall and made more
robust

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

peetersmark, 2025-09-05:
> i renamed it for ya! try it!

I am not sure where to see the changes you made?

But the new script works with my original format, and i am seeing the data on fiftyone now and in the CSV export.

Thanks, this is working great now! To bad I am so late in the year for Michigan, I am not seeing many cute moths and bugs. So far the shed is the best place to see them.

quitmeyer, 2025-09-05:
Ah weird maybe my file didn't save

I basically just changed the entries underneath each of those columns to
match exactly like "kzoo" instead of Kansaszoo or whatever :)

__

Sincerely,
Dr. Andrew James Quitmeyer
Director of Digital Naturalism Laboratories
Affiliate Professor, University of Washington - Human Centered Design and
Engineering
www.dinalab.net

peetersmark, 2025-09-06:
OH! so the **area** and **point** in the CSV file need to match the **project** and **site** of the deployment name? I must have missed that connection.