Digital-Naturalism-Laboratories/Mothbox #79: Mothbot_UI.py hangs on Pick Deployment Folder

peetersmark, 2025-09-17:
**Describe the bug**
running Mothbot_UI.py opens a browser window, when clicking the button "Pick Deployment Folder" it just begins counting seconds and "processing". The other tabs allow drag and drop for files, but this first tab does not allow for any inputs. It just hangs.

**To Reproduce**
running Mothbot_UI.py on win10, in virtual studio code.

**Expected behavior**
allow user to choose deployment folder.

**Additional context**
current files as of 2025-09-17 noon EST

peetersmark, 2025-09-17:
thanks for all the hard work you are doing!
Since all six of these scripts call same or similar paths is there a way to have them reference a single file that could hold the inputs. Then we would only need to change that one file. I am thinking it should contain:

> INPUT_PATH
> METADATA_PATH
> UTC_OFFSET
> TAXA_LIST_PATH
> YOLO_MODEL

I think those are the only parts of the script we need to change for our local environments.

quitmeyer, 2025-09-18:
Hey @peetersmark , it should not hang now when you choose a folder! We tested on pc and mac!

peetersmark, 2025-09-18:
oh very cool, I was able to select my folder and I was able to follow along until "create dataset"
at which point I got the follow error:

> ---📋 Creating Dataset for D:\mothbox\DATA\USA\USA_kzoo_506BkYardMaple_yearlyBrema_2025-09-14\2025-09-09 ---
> scanning for metadata matches... (folder variants: {'kzoo_506bkyardmaple_yearlybrema_2025-09-14', 'usa_kzoo_506bkyardmaple_yearlybrema_2025-09-14'})
> Traceback (most recent call last):
>   File "D:\mark_igloo\mark\Documents\GitHub\Mothbox\AI\Mothbot\Mothbot_CreateDataset.py", line 1138, in <module>
>     metadata= find_csv_match(INPUT_PATH, METADATA_PATH)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "D:\mark_igloo\mark\Documents\GitHub\Mothbox\AI\Mothbot\Mothbot_CreateDataset.py", line 242, in find_csv_match
>     print(f"\u2705 Matched deployment_name = '{matches[0].get('deployment_name')}'")
>   File "D:\mark_igloo\mark\AppData\Local\Programs\Python\Python311\Lib\encodings\cp1252.py", line 19, in encode
>     return codecs.charmap_encode(input,self.errors,encoding_table)[0]
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> UnicodeEncodeError: 'charmap' codec can't encode character '\u2705' in position 0: character maps to <undefined>
> 
> ❌ Create dataset for D:\mothbox\DATA\USA\USA_kzoo_506BkYardMaple_yearlyBrema_2025-09-14\2025-09-09 exited with error code 1
> ------ CreateDataset processing finished ------

I looked up character '\u2705' which is:
Unicode for the White Heavy Check Mark emoji (✅)
But where would it be seeing that?

quitmeyer, 2025-09-18:
Amazing work! thanks for testing that! 

Ill chase down that encoding error and try to fix it or add safeguards!

quitmeyer, 2025-09-18:
By the way, that checkmark comes from the output logs when running the different functions:

```            
                   output_log += f"\n❌ Identification for {folder} exited with error code {process.returncode}\n"
            else:
                output_log += f"✅ Identification completed for {folder}\n"
```

but what's odd is that those similar type of comments run on each function, so i would have expected it to have that error during the detection or ID too

do you see green checks in the console in the GUI when you run the other functions?

quitmeyer, 2025-09-19:
Hey good news! Check out the latest commit for a way better UI that hopefully makes more sense, and I just tested the whole processing path and it seems to work!!

Feel free to try out the whole process @Hubertszcz @briannaljohns 

im gonna close this issue since it seems like we squashed the indefinite hanging bug of the choose folder thing.

closed 2025-09-19 by quitmeyer